gnuton/asuswrt-merlin.ng #725: RT-AX92U: AIMesh mode calling back home

rochmind, 2025-01-01:
**Router Model Affected**
Models: RT-AX92U

**Firmware Version Affected**
3004.388.88

**Is this bug present in upstream Merlin releases too?**
Yes

**Describe the bug**
Recently got fibre to the front of the house but needed another repeater for the back as the signals are weak. Been using RT-AX92U as main AP so got another one as AIMesh mode, however, noticed it is making calls back to:
baidu.com
routerfeedback.asus.com

The other RT-AX92U as AP does not seem to do this - is there a config setting for this?

**To Reproduce**
Steps to reproduce the behavior:
1. Go to LAN
2. Change DNS server to 'pihole' instance or similar DNS lookup setup. 
3. Logs would show the calls back home
4. Asus AIProtection, Adaptive QoS, AICloud, Game, etc are all switched off as well as firmware check

**Expected behavior**
Calling back home is not expected

**Screenshots**
If applicable, add screenshots to help explain your problem or remove this section

rochmind, 2025-01-04:
found nvram setting dns_probe_host was set to www.baidu.com and have set this to null but still goes back to baidu.. even thought dns_probe is set to 0 as well (got this router from eBay and it may have been flashed with with koolshare merlin before)

rochmind, 2025-01-04:
spent another couple of hours on this, enabling dual WAN (don't have another one), then disabling on baidu.com and setting the nvram values below stopped the probing every minute to nothing?!
dns_probe=1
dns_probe_content=*
dns_probe_host=dns.msftncsi.com
dns_probe_timeout=8

https://github.com/gnuton/asuswrt-merlin.ng/blob/master/release/src/router/shared/defaults.c#L2205

closed 2025-01-04 by rochmind

darcyrush, 2025-01-18:
Are you sure its not related to recent malware? https://www.snbforums.com/threads/malware-damaging-asus-routers.92752/

It's hard to find exact information, but it seems it affects the AX- range of routers with certain configurations.

rochmind, 2025-01-28:
i've recently changed the dns_probe_host=1.1.1.1 and dns_probe_content=1.1.1.1 as well as applied the latest gnuton firmware from 2 weeks ago.

and no traffic in pihole>>unbound>>nextdns (my setup) logs for the aimesh node for the probe