gnuton/asuswrt-merlin.ng #727: CakeQOS not limiting download

eConnah, 2025-01-08:
**Router Model Affected**
Models: TUF-AX3000 V2 

**Firmware Version Affected**
3004.388.8_2-gnuton1

**Is this bug present in upstream Merlin releases too?**
Unsure

**Describe the bug**
The bandwidth limiter for upload works perfectly well but for download it just doesn't work. It works using flex but I'm unsure why it doesn't with cake. I'm using pppoe on full fibre, 100/20 connection.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to adaptive qos -> qos.
2. Click on cake.
3. Scroll down to manual bandwidth, enable set limits.
4. Run speedtest.
5. See download limit is not working.

**Expected behavior**
Download limit should work.

**Screenshots**
[image]
[image]
[image]
[image]

**Fix from dave14305**
`tc filter add dev ppp0 parent ffff: protocol all prio 10 u32 match u32 0 0 flowid 1:1 action mirred egress redirect dev ifb4ppp0`

Link: https://www.snbforums.com/threads/cakeqos-not-limiting-download.93504/

dave14305, 2025-01-08:
`CONFIG_NET_CLS_MATCHALL` needs to be enabled for 4.19 kernel devices.

gquittet, 2025-06-25:
This workaround does not work anymore with latest [3004.388.9_2-gnuton1](https://github.com/gnuton/asuswrt-merlin.ng/releases/tag/3004.388.9_2-gnuton1) release.

So I rollback to [3004.388.8_4-gnuton1](https://github.com/gnuton/asuswrt-merlin.ng/releases/tag/3004.388.8_4-gnuton1)

alfky12, 2025-06-27:
Hi @dave14305 can you help out again?

dave14305, 2025-06-27:
I don’t know what broke.

gnuton, 2025-07-17:
I have enabled again CONFIG_NET_CLS_MATCHALL  fix here e19fe6df05

eConnah, 2025-07-24:
I installed it and tested it with a speedtest and this time neither the upload or the download got limited, I set a manual limit much lower than my isp provides and it was way over that limit. I tried the previous temporary fix
`tc filter add dev ppp0 parent ffff: protocol all prio 10 u32 match u32 0 0 flowid 1:1 action mirred egress redirect dev ifb4ppp0`
however that is no longer a recognized command. I don't really know anything about the topic but if you need any other info just ask.

dave14305, 2025-07-25:
> if you need any other info just ask.

What’s in `/tmp/qos` file after installing this build?

eConnah, 2025-07-26:
This is the output with cake and a manual limit set if that matters.
```
#!/bin/sh
source /etc/cake-qos.conf

case "$1" in
start)
# Upload
	tc qdisc add dev $ULIF root cake $ULPRIOQUEUE $ULBW $OVERHEAD $FRAMING $ULOPTIONS 2>/dev/null

# Download
	ip link add name $MIF type ifb 2>/dev/null
	tc qdisc add dev $DLIF handle ffff: ingress 2>/dev/null
	tc qdisc add dev $MIF root cake $DLPRIOQUEUE $DLBW $OVERHEAD $FRAMING $DLOPTIONS 2>/dev/null
	ip link set $MIF up 2>/dev/null
	tc filter add dev $DLIF parent ffff: prio 10 matchall action mirred egress redirect dev $MIF 2>/dev/null

	;;
stop)
	tc qdisc del dev $ULIF root 2>/dev/null
	tc qdisc del dev $DLIF ingress 2>/dev/null
	tc qdisc del dev $MIF root 2>/dev/null
	ip link set $MIF down 2>/dev/null
	ip link del dev $MIF 2>/dev/null
	;;
*)
esac
```

dave14305, 2025-07-26:
> This is the output with cake and a manual limit set if that matters.

Can you share `/etc/cake-qos.conf` also?

eConnah, 2025-07-26:
```
#!/bin/sh

ULIF='ppp0'
DLIF='ppp0'
MIF='ifb4ppp0'
ULBW='bandwidth 18432kbit'
DLBW='bandwidth 95232kbit'
OVERHEAD='overhead 0 mpu 0'
FRAMING=''
ULPRIOQUEUE='diffserv3'
DLPRIOQUEUE='besteffort'
ULOPTIONS='nat dual-srchost'
DLOPTIONS='nat wash dual-dsthost ingress'
```

dave14305, 2025-07-26:
Ok, the next step would be to run these commands one by one and look for errors.
```
tc qdisc
source /etc/cake-qos.conf
tc qdisc add dev $ULIF root cake $ULPRIOQUEUE $ULBW $OVERHEAD $FRAMING $ULOPTIONS
ip link add name $MIF type ifb
tc qdisc add dev $DLIF handle ffff: ingress
tc qdisc add dev $MIF root cake $DLPRIOQUEUE $DLBW $OVERHEAD $FRAMING $DLOPTIONS
ip link set $MIF up
tc filter add dev $DLIF parent ffff: prio 10 matchall action mirred egress redirect dev $MIF
tc qdisc
```

eConnah, 2025-07-26:
a@a:/tmp/home/root# tc qdisc add dev $ULIF root cake $ULPRIOQUEUE $ULBW $OVERHEAD $FRAMING $ULOPTIONS
RTNETLINK answers: No such file or directory
This one gave an error

eConnah, 2025-07-26:
all commands
```
a@a:/tmp/home/root# tc qdisc
qdisc noqueue 0: dev lo root refcnt 2 
qdisc noqueue 0: dev archer root refcnt 2 
qdisc pfifo_fast 0: dev eth1 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev eth0 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev eth2 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev eth3 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev eth4 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev spu_us_dummy root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev spu_ds_dummy root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev eth5 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev eth6 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc noqueue 0: dev br0 root refcnt 2 
qdisc pfifo_fast 0: dev ifb4ppp0 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev ppp0 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc ingress ffff: dev ppp0 parent ffff:fff1 ---------------- 
a@a:/tmp/home/root# source /etc/cake-qos.conf
a@a:/tmp/home/root# tc qdisc add dev $ULIF root cake $ULPRIOQUEUE $ULBW $OVERHEAD $FRAMING $ULOPTIONS
RTNETLINK answers: No such file or directory
a@a:/tmp/home/root# ip link add name $MIF type ifb
RTNETLINK answers: File exists
a@a:/tmp/home/root# tc qdisc add dev $DLIF handle ffff: ingress
RTNETLINK answers: File exists
a@a:/tmp/home/root# tc qdisc add dev $MIF root cake $DLPRIOQUEUE $DLBW $OVERHEAD $FRAMING $DLOPTIONS
RTNETLINK answers: No such file or directory
a@a:/tmp/home/root# ip link set $MIF up
a@a:/tmp/home/root# tc filter add dev $DLIF parent ffff: prio 10 matchall action mirred egress redirect dev $MIF
RTNETLINK answers: File exists
We have an error talking to the kernel
a@a:/tmp/home/root# tc qdisc
qdisc noqueue 0: dev lo root refcnt 2 
qdisc noqueue 0: dev archer root refcnt 2 
qdisc pfifo_fast 0: dev eth1 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev eth0 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev eth2 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev eth3 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev eth4 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev spu_us_dummy root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev spu_ds_dummy root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev eth5 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev eth6 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc noqueue 0: dev br0 root refcnt 2 
qdisc pfifo_fast 0: dev ifb4ppp0 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev ppp0 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc ingress ffff: dev ppp0 parent ffff:fff1 ---------------- 
a@a:/tmp/home/root# 
```

dave14305, 2025-07-26:
It looks as if CAKE module is no longer built into the kernel anymore. @gnuton should verify if CONFIG_NET_SCH_CAKE is set. Do you see a module file anywhere?
`find / -name sch_cake*`

dghodgson, 2026-01-05:
This is still an issue in 3004.388.10_2-gnuton1

`find / -name sch_cake*` returns nothing, and I cannot seem to disable Archer hardware acceleration. The nvram vars are set to disable, but after committing and rebooting the system info page still shows it's enabled

alfky12, 2026-01-06:
Please try my build of the firmware based on Gnuton. This is the tuf version. It's only for the TUF-AX3000_V2.
https://github.com/alfky12/tuf-asuswrt-merlin.ng/actions/runs/20730784723/artifacts/5030748184

m0wseu, 2026-09-21:
Update everyone. I have a build with the latest gnuton release 388.11_1 —thanks to his workflow, that fixed sch_cake for TUF-AX3000_V2. Would anyone still interested trying out the firmware (default ui)? Let me know and I'll attached it here. @gquittet @eConnah @dghodgson @alfky12

eConnah, 2026-10-06:
I can no longer test it as I have left my mum's house but if its fixed in latest gnuton release should I close the issue?