gnuton/asuswrt-merlin.ng #729: Update notice to 3004.388.8_4-gnuton1 though unavailable.

gurple, 2025-01-12:
**Router Model Affected**
Models: DSL-AX82U

**Firmware Version Affected**
3004.388.8_4-gnuton0_alpha1

**Is this bug present in upstream Merlin releases too?**
No

**Describe the bug**
The router is advertising the availability 3004.388.8_4-gnuton1 though it is unable to acquire the firmware image for download.

LiroyvH, 2025-01-12:
I see the same from the router interface of the AX3000_v2. It is linking to "https://gnuton.github.io/asuswrt-merlin.ng/updates/TUF-AX3000_V2" of which GitHub says it 404's.

gnuton, 2025-01-14:
this should be fixed now, correcT?

gurple, 2025-01-14:
Yes, it appears to be fixed. And the update to the now-available 3004.388.8_4-gnuton1 firmware is complete. So far no issues.
Thanks and cheers!

closed 2025-01-14 by gurple