gnuton/asuswrt-merlin.ng #733: RT-AX82U - Firmware from 3004.388.8_2-gnuton1will not apply 3004.388.8_4-gnuton1

raykai, 2025-01-17:
**Router Model Affected**
Models: RT-AX82U

**Firmware Version Affected**
3004.388.8_4-gnuton1

**Is this bug present in upstream Merlin releases too?**
Not applicable as Merlin only does the PRO model.

**Describe the bug**
updated to 3004.388.8_2-gnuton1 firmware fine a few months a go now trying to update to 3004.388.8_4-gnuton1 but fails.

**Expected behavior**
[image]

it should work and not fail

gnuton, 2025-01-17:
Try to flash stock firmware first then merlin. doest that work that way?
(take a backup before)

raykai, 2025-01-17:
flash stock 3.0.0.4.388_25017 firmware, that worked. Then I upload the 3004.388.8_4-gnuton1 fail.

[image]

gnuton, 2025-01-17:
Can you please tell me the hw version you have? 
RT-AX82 has V1 and V2 firmware variants.

raykai, 2025-01-17:
V1

gnuton, 2025-01-17:
please try to flash the firmware once more and show me the logs.

raykai, 2025-01-17:
Il do when I get up, as for the  pre-release I did not try.

raykai, 2025-01-17:
what's the proper way to get the logs that you need?

gnuton, 2025-01-19:
the logs from the web ui are enough