gnuton/asuswrt-merlin.ng #738: 386.14 release ready for out-of-beta?

mackonsti, 2025-01-18:
Hello @gnuton happy new year!

Any idea if this release is ready for prime-time and be officially released?
https://github.com/gnuton/asuswrt-merlin.ng/releases/tag/386.14_0-gnuton0_beta1

Grazie mille -- Konsti

gnuton, 2025-01-19:
Hi yesterday I merged 14_2

mackonsti, 2025-01-21:
Thank you @gnuton for always remembering the old DSL-AC68U users of 386.xx builds! 🙏 
Looking forward to it.

mackonsti, 2025-02-12:
Hello @gnuton ! Any chance for a stable release for 386.x and DSL-AC68U users, planned in the coming days, since you said you have merged 14_2 ? Grazie!

mackonsti, 2025-06-07:
Replied in https://github.com/gnuton/asuswrt-merlin.ng/issues/767#issuecomment-2946206265 hopefully we will get a last ever build to use on this 386.14 build.

closed 2025-06-07 by mackonsti