gnuton/asuswrt-merlin.ng #739: AX58U V2 3004.388.8_2 to 8_4 not updating

cseryz, 2025-01-19:
**Router Model Affected
Ax58u v2

**Firmware Version Affected**
3004.388.8_2-gnuton1

**Is this bug present in upstream Merlin releases too?**
No

**Describe the bug**
firmware update doesnt work to 3004.388.8_4-gnuton1

**To Reproduce**
Steps to reproduce the behavior:
Manual fw update, or "auto" fw update

**Expected behavior**
working update

**Screenshots**
If applicable, add screenshots to help explain your problem or remove this section

After update firmware, router reboots but old version of fw still on place.

Blarglick, 2025-01-26:
Having the exact same issue with a TUF-5400.

shnoah641, 2025-01-27:
Please use Firmware rescue tool on your routers model page in the drivers and tools and use that utility to install it I had the same issue also follow the rescue instructions

Blarglick, 2025-02-08:
@shnoah641 Thank you for the rescue tool information. It didn't work, the router was entering rescue mode looking at power light, but software would not recognise it. 

I was able to flash the new firmware by hard resetting the router to factory defaults (WPS button method) and then flash the new firmware via the web interface,

shnoah641, 2025-02-08:
> [@shnoah641](https://github.com/shnoah641) Thank you for the rescue tool information. It didn't work, the router was entering rescue mode looking at power light, but software would not recognise [...]

Ok good glad to hear and yeah the asus recovery utility is kind of funky like that you have to hold the reset button as you restart the router until the lower light starts to blink then that indicates it's in recovery mode.

Blarglick, 2025-02-08:
@shnoah641 

Yes, I did that multiple times. I had a continuous ping running to watch the router to make sure the program could have seen it. Did exactly per the site instructions. I could see the light starting to blink, and ping indicated it was up, TUF light was also red flashing slow along with the power light. But the recovery software would always say the router was not in recovery mode. 

Tried varying lengths of holding down the reset but to no avail. Power light would start flashing, but software just would not work. 

May not work with TUF-AX5400, or my firmware was borked in a way that it couldn't enter rescue mode properly. 

Thanks for the info, as it did put me on the road to fixing it (Hard factory reset in my case)

syskin345, 2025-02-09:
So I had the exact same symptoms on RT-AX56U and in the end, I rebooted the router and immediately applied the update one more time and then it worked.

cseryz, 2025-03-19:
Out of curiosity tried again today, manual update from laptop.
It did worked ...
No idea, i changed nothing since.
:)

closed 2025-03-19 by cseryz