gnuton/asuswrt-merlin.ng #747: RT-AX58U V2 LED lights

Andy-Shiekh, 2025-02-10:
**Router Model Affected**
Models: (eg: RT-AX58U V2)

**Firmware Version Affected**
3004.388.8_4-gnuton1

**Is this bug present in upstream Merlin releases too?**
There is no Merlin release for RT-AX58U V2

**Describe the bug**
Front LED lights don't run, but some come on during boot on ASUS firmware
so may be a hardware issue

jacklul, 2025-02-15:
This might be a hardware issue because I noticed the same thing on stock firmware, seen similar threads about it on reddit as well for V2 specifically.

akahomebody, 2025-03-09:
I'm having the same issue. Some of the LEDs were running when on stock (not all though), but after flashing the latest stable merlin firmware none of the lights work. Curious!

Andy-Shiekh, 2025-03-09:
As far as I know there is no Merlin version for the RT-AX58U V2

akahomebody, 2025-03-09:
Ah, I meant the gnuton firmware. I assumed it was a fork of Merlin.

Andy-Shiekh, 2025-03-09:
Ah, my bad.

SultanChoudhury, 2025-05-21:
> I'm having the same issue. Some of the LEDs were running when on stock (not all though), but after flashing the latest stable merlin firmware none of the lights work. Curious!

Same here.