gnuton/asuswrt-merlin.ng #751: USB drives not recognized after rrboot

McNessie, 2025-02-17:
**Router Model Affected**
Models: RT-AX82U

**Firmware Version Affected**
388.8_4-gnuton1

**Is this bug present in upstream Merlin releases too?**
Not sure

**Describe the bug**
Drives connected to the USB port are not recognized after reboot of router

**To Reproduce**
Steps to reproduce the behaviour:
1. Connect USB drive and make sure it's mounted
2. reboot router
3. Check to see if USB drive still connected
4. It will say No device connected

**Expected behavior**
I expect the drive to be connected and auto mounted

gnuton, 2025-02-28:
any log to see?

gnuton, 2025-03-02:
```
blk_update_request: I/O error, dev sda, sector 0
Buffer I/O error on dev sda1, logical block 3702784, lost sync page write
```

These indicate physical read/write failures on the disk, which suggests potential hardware problems (e.g., bad sectors or failing storage media).

Cache synchronization failure:
`
sd 0:0:0:0: [sda] Synchronize Cache(10) failed: Result: hostbyte=DID_NO_CONNECT driverbyte=DRIVER_OK`
This suggests a failure to synchronize the cache, which could mean:
The device was disconnected improperly (e.g., a USB drive being unplugged).
Power issues affecting the drive (especially on USB-connected storage).
Cable or port problems causing unstable communication.

There is high chance this is an hardware issue