gnuton/asuswrt-merlin.ng #753: RT-AX82U V2 cfg_server memory leak.

pymumu, 2025-03-03:
**Router Model Affected**
RT-AX82U V2

**Firmware Version Affected**
3004.388.8_4-gnuton0_alpha1

**Is this bug present in upstream Merlin releases too?**
Not tested

**Describe the bug**
The router has been running for 16 days and the memory of cfg_server is slowly growing.

**To Reproduce**
None

**Expected behavior**
No memory leak

**Screenshots**
[image]

pymumu, 2025-03-03:
2 hours after:
```
Load average: 0.74 0.68 0.71 2/156 1261
  PID  PPID USER     STAT   VSZ %VSZ CPU %CPU COMMAND
 2877     1 admin    S    77112 15.1   0  0.1 cfg_server
 2854     1 admin    S    20176  3.9   0  0.0 conn_diag
22457     1 admin    S    19484  3.8   0  0.0 aaews --sdk_log_dir=/tmp
 2851     1 admin    S    18636  3.6   0  0.0 roamast
 2873  2854 admin    S    17612  3.4   1  0.0 amas_portstatus
 2638     1 admin    S    14536  2.8   0  0.0 /sbin/netool
 2834     1 admin    S    14512  2.8   1  0.1 networkmap --bootwait
 2808     1 admin    S    14448  2.8   0  0.0 ahs
    1     0 admin    S    13876  2.7   1  0.0 /sbin/init
 2660     1 admin    S    13488  2.6   1  0.1 nt_center
 2635     1 admin    S    13456  2.6   0  0.0 nt_monitor
```

process status
```
admin@RT-AX82U-A510:/tmp/home/root# cat /proc/2877/status
Name:   cfg_server
Umask:  0000
State:  S (sleeping)
Tgid:   2877
Ngid:   0
Pid:    2877
PPid:   1
TracerPid:      0
Uid:    0       0       0       0
Gid:    0       0       0       0
FDSize: 32
Groups:
VmPeak:    78136 kB
VmSize:    77112 kB
VmLck:         0 kB
VmPin:         0 kB
VmHWM:     63076 kB
VmRSS:     63068 kB
RssAnon:           59228 kB
RssFile:            3828 kB
RssShmem:             12 kB
VmData:    65652 kB
VmStk:       132 kB
VmExe:       600 kB
VmLib:      6160 kB
VmPTE:        84 kB
VmSwap:        0 kB
CoreDumping:    0
```

gnuton, 2025-03-04:
There is nothing I can do for cfg server since it's a pre built binary from Asus. 
Moreover it s a central service so its memory grows with its usage.
Increase mem over time doesn't mean there is a memory leak. 
Feel free to open this thread when you get an OOM

closed 2025-03-04 by gnuton