gnuton/asuswrt-merlin.ng #757: RT-AX82U v1 unable to revert to stock image

Amoeba00, 2025-03-10:
**Router Model Affected**
RT-AX82U v1

**Firmware Version Affected**
3004.388.8_4-gnuton1

**Is this bug present in upstream Merlin releases too?**
N/A

**Describe the bug**
Tried to install the[ latest stock firmware from Asus](https://www.asus.com/us/networking-iot-servers/wifi-routers/asus-gaming-routers/rt-ax82u/helpdesk_bios?model2Name=RT-AX82U) - get the "firmware upgrade unsuccessful" error.
Tried the one from June, October, November, and both recent ones from February.

**To Reproduce**
1. Navigate to the router upgrade page
http://www.asusrouter.com/Advanced_FirmwareUpgrade_Content.asp
2. Click on the upload link
3. Select the firmware file
4. Receive error message, log back into router

**Expected behavior**
Firmware upgrade successful

**Screenshots**
(Same ones others have posted here)

HarryDotMYx, 2025-03-12:
Do you already try put your router into Rescue Mode / Emergency Firmware Restoration?

Amoeba00, 2025-03-12:
I have not.  I got it down to v8_2 and was just waiting until someone could offer guidance on next steps.

HarryDotMYx, 2025-03-13:
> I have not. I got it down to v8_2 and was just waiting until someone could offer guidance on next steps.

https://www.youtube.com/watch?v=T8YKnpXxhx0 - Here the video for Restoration Firmware. 

https://www.asus.com/networking-iot-servers/wifi-routers/asus-gaming-routers/rt-ax82u/helpdesk_bios?model2Name=RT-AX82U - The Original Firmware from Asus.

https://www.asus.com/my/support/faq/1000814/ - Here the documentation. 

Still didn't clear? then comment here ;)

Amoeba00, 2025-03-13:
So then it's not a bug with this merlin software and under normal circumstances I should just be able to downgrade my v1 device via normal browser method?

Thanks for the links - will report back results.

HarryDotMYx, 2025-03-13:
Yes, it’s not a bug with the Merlin software. Under normal circumstances, you should be able to downgrade/upgrade your router using the Firmware Restoration method.

Amoeba00, 2025-03-18:
> Yes, it’s not a bug with the Merlin software. Under normal circumstances, you should be able to downgrade/upgrade your router using the Firmware Restoration method.

Oh - so using the Firmware Restoration method is the correct way to downgrade and not via normal firmware page in Merlin? I could have sworn I read somewhere that you could just download by normal means.

HarryDotMYx, 2025-03-18:
> Oh - so using the Firmware Restoration method is the correct way to downgrade and not via normal firmware page in Merlin? I could have sworn I read somewhere that you could just download by normal [...]

Yes, the Firmware Restoration Tool is the recommended method for downgrading from Asuswrt-Merlin back to the official ASUS stock firmware, especially if the normal firmware upgrade method fails.

Amoeba00, 2025-03-19:
Just wanted to report that the Restoration Utility worked and all is well.  May revisit this Merlin fork at a later date.  Thanks much - Cheers!

HarryDotMYx, 2025-03-19:
Niceee! You welcome ;)

HarryDotMYx, 2025-03-19:
@Amoeba00 @gnuton can close this issues

closed 2025-03-20 by Amoeba00