gnuton/asuswrt-merlin.ng #758: Not getting WAN IP

Randi1115, 2025-03-13:
**Router Model Affected**
Models: (eg: DSL-AX82U)

**Firmware Version Affected**
3004.388.8_4-gnuton1

**Is this bug present in upstream Merlin releases too?**
No

**Describe the bug**
With this firmware version the router works for about one week and then it loses the WAN connection. I'm connecting to internet via FTTP connection on this router. When logged on to the router the error is "Your ISP's DHCP does not function correctly." Rebooting the router does work only for few minutes and it's the same message again. I've reset it to factory settings and redone all settings from scratch but it's still the same. I've even done a hard reset on the router but it's still the same.

If I plug in ISPs router there's no issue. I've also checked with the ISP and there's no issue on the line. All lights on the ONT are green. If I use the stock firmware then there's no issue either. The WAN connection stays on for over 2 weeks without any issues. I've reverted to the stock firmware for now.

**To Reproduce**
Steps to reproduce the behavior:
1. Install this version of Firmware
2. Run for about a week

**Expected behavior**
Router shouldn't lose WAN connection this frequently.

HarryDotMYx, 2025-03-14:
What ISP that you use?

Randi1115, 2025-03-16:
Hi,

The ISP is TalkTalk in UK.

Please see the log file attached.

[WAN-Error Log.txt](https://github.com/user-attachments/files/19272739/WAN-Error.Log.txt)