gnuton/asuswrt-merlin.ng #762: GT-BE19000

Packus, 2025-04-07:
Hi!
Question about GT-BE19000, do you plan to support this model?

unisoftenterprises, 2025-04-29:
Merlin supports this unit. Firmware available right now and using newer codebase.

emeraquino, 2025-04-29:
Not sure where you got the info from @unisoftenterprises but I don't see **GT-BE19000** on the [official supported devices list yet](https://github.com/RMerl/asuswrt-merlin.ng/wiki/Supported-Devices).

Note that [GT-BE98_PRO](https://rog.asus.com/networking/rog-rapture-gt-be98-pro/spec/) is a similar model and is supported but still not the same as the [GT-BE19000](https://rog.asus.com/de/networking/rog-rapture-gt-be19000/spec/)

unisoftenterprises, 2025-04-29:
It's the GT-BE96U aka GT-BE19000.....

gnuton, 2025-05-02:
please do not open ticket for model support requests.

closed 2025-05-02 by gnuton