gnuton/asuswrt-merlin.ng #763: Wireless nodes drop off

mstendah, 2025-04-16:
**Router Model Affected**
Models: XT8
**Firmware Version Affected**
388.8_4

**Is this bug present in upstream Merlin releases too?**
Not sure

**Describe the bug**
From time to time my wireless XT8 Mesh nodes are lost and need yo be reset and added back. This never happened with ASUS XT8 stock firmware

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
I would expect the nodes to stay connected or recover automatically 

**Screenshots**
If applicable, add screenshots to help explain your problem or remove this section

KrekkieD, 2025-04-17:
I am experiencing the same problem. It happens (at least) when the main node reboots. All my mesh nodes end up with a red light (wired backhaul only, no connection).

Web interface does not show the mesh nodes, but interestingly it does still see them on the clients list.

The web interface also returns errors when trying to open the clients list. The json response for `/appGet.cgi?hook=get_cfg_clientlist()` seems malformed:
```
{ "get_cfg_clientlist": }
```

I was honestly starting to think there was a corrupt bit of storage somewhere on the main node, as no one else had reported this problem. "glad" to see that may not be the case.

KrekkieD, 2025-04-17:
Restoring backed up config file or JFFS file does not fix it, unfortunately.

hulleyrob, 2025-04-21:
unfortunately although the latest [3.0.0.4.388_24709] stock firmware is stable the wifi is dog slow, maybe this is only in AP mode but its massively slower. however had no disconnections unlike on the previous gntuon firmware.

mstendah, 2025-04-21:
What is slow to you, I get over 800Mbps via WiFi using my iPhone 16Pro, which is pretty much the same as over my Wired connection. And this is on Stock firmware.

[Image.png]

hulleyrob, 2025-04-21:
try 40Mbps...

what's interesting is the PHY is in the 800 range yet im lucky to hit 40-50 transferring data

[image]