gnuton/asuswrt-merlin.ng #767: Will DSL-AC68U still be supported?

colibris79, 2025-05-07:
Hi @gnuton , 
One quick question, after RT-AC68U not being supported anymore for Merlin (last version RT-AC68U_386.14_2), will you still make any forks for DSL-AC68U?
Thanks

mackonsti, 2025-06-02:
Hi @gnuton thank you again for breathing life to our old DSL-AC68U routers, when ASUS abandoned them. Little life is left on mine too, with optical fibre coming soon in my town...

If you have the _last ever_ **stable** build to test openly for people (386_14.0) perhaps you can provide it with ⚠ warning signs ⚠ for those wishing one more last build to test and use, explaining that support has ended.

It's a shame to not release this build as it's ready, I think. Or name it RC (Release Candidate) if you think it will help people 😆 

Thank you again for everything!

mackonsti, 2025-06-07:
Thanks @gnuton looking forward to this last stable release!

colibris79, 2025-06-26:
That would be great. My AC68U is still working fine and would leave it with that firmware as a backup router

Ambrog, 2025-07-11:
daje! :3

ZsGabor, 2025-08-28:
> MAke sense. I have a stable release but I didn't have any tester and I left it there because no one was interested in this.

Hi Antonio!
I'm not sure what the testing means exactly - what steps or processes should be tested or done -, but I am happy to support You to create a final firmware release of this router.
G.

ZsGabor, 2025-08-28:
> MAke sense. I have a stable release but I didn't have any tester and I left it there because no one was interested in this.

Hi Antonio!
I'm not sure what the testing means exactly - what steps or processes should be tested or done -, but I am happy to support You, @gnuton to create a final firmware release of this router.
G.