gnuton/asuswrt-merlin.ng #784: Can't Add Static Route to Guest Network

andrewk123, 2025-06-02:
**Router Model Affected**
Models: XT8

**Firmware Version Affected**
3004.388.8_4-gnuton1

**Is this bug present in upstream Merlin releases too?**
Yes

**Describe the bug**
If I add a static route using the web gui to an IP address on the “Non Guest” part of the router (say 192.168.48.4) then every thing works normally. A route is created in the gui and a matching entry created in the route table.  If however I point a static route at a device in the guest 1 network (in this case 192.168.103.250) the route appears to be created in the gui (see attached) however looking at route -n no route is actually created and a bunch of the following error is generated.

Jun 2 06:54:07 kernel: ^[[0;33;41m[ERROR mcast] bcm_mcast_netlink_process_snoop_cfg,1015: interface 24 could not be found^[[0m
Jun 2 06:54:07 kernel: ^[[0;33;41m[ERROR mcast] bcm_mcast_netlink_process_snoop_cfg,1015: interface 24 could not be found^[[0m
Jun 2 06:54:07 kernel: ^[[0;33;41m[ERROR mcast] bcm_mcast_netlink_process_snoop_cfg,1015: interface 24 could not be found^[[0m

If however, I delete the static route from the gui and add it via ssh with:

ip route add 192.168.20.0/24 via 192.168.103.250

The route is created correctly and everything works correctly.  As expected the web gui is “unaware” of this route (even though it works just fine).

Here is a snippet from the route table (route -n) (note the 81.0 subnet is created by OpenVPN):

192.168.20.0    192.168.103.250 255.255.255.0   UG    0      0        0 br3
192.168.48.0    0.0.0.0         255.255.252.0   U     0      0        0 br0
192.168.81.0    0.0.0.0         255.255.255.0   U     0      0        0 tun21
192.168.102.0   0.0.0.0         255.255.255.0   U     0      0        0 br2
192.168.103.0   0.0.0.0         255.255.255.0   U     0      0        0 br3

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'LAN->Route'
2. Enable "Static Routes"
3. Add Static Route Details and Click on '+'
4. Click on 'Apply'

**Expected behavior**
Route is added as per ssl example.

**Screenshots**
Failed Route Addition

[image]

gnuton, 2025-06-03:
Hi,
Can you please test the latest beta?

andrewk123, 2025-06-03:
I cannot do so for a couple of weeks since I do not have "spare" hardware available to test with.  As soon as I do I will load up the latest beta and give it a try.

andrewk123, 2025-06-13:
I tried this again in 3004.388.9_2-gnuton0_beta3 as requested.

No change in functionality, the "route" is added on the Web GUI but looking at route -n no route is added.