gnuton/asuswrt-merlin.ng #785: No TUF UI on TUF-AX5400_3004_388.9_2-gnuton0_beta1_tuf

NoodleSoup, 2025-06-02:
**Router Model Affected**
Models: (eg: DSL-AX82U/RT-AX82U/...)
TUF-AX5400

**Firmware Version Affected**
3004.388.9_2-gnuton0_beta1

**Is this bug present in upstream Merlin releases too?**
No

**Describe the bug**
No TUF UI on the TUF version.

**To Reproduce**
Steps to reproduce the behavior:
1. Install the new beta on TUF-AX5400
2. After rebooting there is no TUF UI

**Expected behavior**
TUF UI should be there on the TUF version.

pibesk, 2025-06-05:
same router, same issue but on beta2

gnuton, 2025-06-10:
I guess  fixed it (not tested yet)
Test images: https://github.com/gnuton/asuswrt-merlin.ng/actions/runs/15559304376/job/43807417164