gnuton/asuswrt-merlin.ng #786: UI issue on 3004.388.9_2-gnuton0_beta2

micron888, 2025-06-04:
Current version
**3004.388.9_2-gnuton0_beta2**

Missing Tools

[image]

Game boost page missing menu after a while

[image]

network tools page missing menu

[image]

gnuton, 2025-06-04:
Are you using the TUF or no_tuf ui image?

pibesk, 2025-06-04:
Same router, same version - both pages visible. 

Did beta1/beta2 change UI that much? Looks inconsistent on my roiuter now - game boost have old tuf look. The rest looks like non tuf ui.

micron888, 2025-06-05:
> Are you using the TUF or no_tuf ui image?

Non tuf version

MuJian669, 2025-06-12:
> 3004.388.9_2-gnuton0_beta3 is out with TUF UI

[image]

BETA3 TUF UI homepage is not healthy, network tools page missing menu,Game boost page missing menu.

Beta3 NO TUF UI network homepage is healthy, tools page missing menu, Game boost page missing menu.

pibesk, 2025-06-12:
what is interesting on Beta3 when you enter some url like http://192.168.1.32/Advanced_FirmwareUpgrade_Content.asp

Suddenly page looks ok - so only home page is broken?

[image]

ming019, 2025-07-03:
Similar issue on the stable release [3004.388.9_2-gnuton1]. TUF-AX3000(China's version).