gnuton/asuswrt-merlin.ng #787: RT-AX58U_V2_3004_388.9_2-beta2 Skynet Failed on IPSets and IPTables Rules

dowden20, 2025-06-04:
**Router Model Affected**
Models: RT-AX58U_V2

**Firmware Version Affected**
3004_388.9_2 - beta2

**Is this bug present in upstream Merlin releases too?**
Not sure

**Describe the bug**
Skynet keeps failing after upgrade to 388.9-2
The fix is to rollback to 388.8-4
I have tried uninstall and reinstall Skynet. Used a different USB

**To Reproduce**
Steps to reproduce the behavior:
1. Run amtm, Install Skynet

2. See error
Router Model; RT-AX58U
Skynet Version; v7.6.4 (30/12/2024) (a34f2e22a26a7acea315fd8ea38c4d27)
iptables v1.4.15 - (eth0 @ 192.168.1.1)
ipset v7.6, protocol version: 7
IP Address; (108.202.*.*)
FW Version; 388.9_2-gnuton0_beta2 (Jun 4 2025) (4.19.183)
Install Dir; /tmp/mnt/sda1/skynet (4.9G / 7.4G Space Available)
SWAP File; /tmp/mnt/sda1/myswap.swp (2.0G)

IPSets                              | [Failed]
IPTables Rules                      | [Failed]

**Expected behavior**

Expect both to shows as Passed

**Screenshots**

dowden20, 2025-06-04:
[250604_debug.txt](https://github.com/user-attachments/files/20595829/250604_debug.txt)

gnuton, 2025-06-06:
@dowden20 my guess for now is that some modules are missing like XT_SET, but I do need to check it out

Khayla, 2025-06-22:
I can confirm this is an issue for me too on 3004.388.9_2-gnuton0_beta3

Dobrotin, 2025-06-25:
I can confirm this is an issue for me too on 3004.388.9_2-gnuton1

gnuton, 2025-07-24:
Test images here https://github.com/gnuton/asuswrt-merlin.ng/actions/runs/16487789919
fix: https://github.com/gnuton/asuswrt-merlin.ng/commit/01c0423dd97bfd520083e867bcc9af9ea76e9faa

gnuton, 2025-07-25:
can anyone test the images and report?

Dobrotin, 2025-07-25:
> Test images here https://github.com/gnuton/asuswrt-merlin.ng/actions/runs/16487789919 fix: [01c0423](https://github.com/gnuton/asuswrt-merlin.ng/commit/01c0423dd97bfd520083e867bcc9af9ea76e9faa)

Thank you, all modules are available.

gnuton, 2025-07-29:
cool merged to master

closed 2025-07-29 by gnuton