gnuton/asuswrt-merlin.ng #788: RT-AX58U_V2_3004_388.9_2-beta2 aaews process issues

hammerjoe, 2025-06-06:
**Router Model Affected**
Models: RT-AX82U

**Firmware Version Affected**
388.9 beta 2

**Is this bug present in upstream Merlin releases too?**
Dont know

**Describe the bug**
A clear and concise description of what the bug is.
Everyday have connection issues. Put the system log into chatgpt and it reports that the process aaews is crashing repeatedly. The crash is a segmentation fault (signal 11) where PC (program counter) is 0x0 - meaning it tried to execute a null pointer.
Its happening in multiple cpus (0, 1) and across different pids (ex 3027, 3032, 3064)

Aaews is a background service/process on asus routers and it's likely related to aiprotection, aimesh, adaptive qos.
Possibly analytics, clud sync, or monitoring modules.

hammerjoe, 2025-06-07:
More details from Deepseek this time:
Key Problem: Recurring Kernel Crashes (Fatal Signal 11)
Error Type: fatal signal 11 (SIGSEGV - Segmentation Fault)

Affected Process: aaews (ASUS proprietary service, likely related to web services/cloud connectivity)

Frequency: Crashes occur every 12 seconds (22:18:09, 22:18:21, 22:18:33, etc.)

Kernel Version: Linux 4.1.52 (custom ASUS build)

Technical Breakdown
Crash Cause:

The process aaews is attempting to access invalid memory addresses:

PC is at 0x0 (Null pointer dereference)

PC is at 0xb65db170 (Invalid/corrupted memory address)

This triggers a segmentation fault, forcing the kernel to terminate the process.

Process Details:

PID: Varies (6819, 6824, 6842, etc.), indicating repeated crashes/restarts.

Comm: aaews (ASUS AAE Web Service - handles remote management/cloud features).

CPU: Crashes occur on both CPU cores (0 and 1).

Stack Trace Analysis:

The crash originates in user-space (Mode USER_32), not kernel code.

Call stack shows:

text
[<c0026f20>] (unwind_backtrace)
[<c0022cf8>] (show_stack)
[<c045ba6c>] (dump_stack)
[<c003ab8c>] (get_signal)  <-- Signal 11 handled here
This confirms the issue is in the aaews application, not the kernel.

Hardware Context:

ISA ARM: Router uses an ARM-based SoC (common for ASUS routers).

No hardware errors reported, suggesting a software/driver issue.

Root Cause Assessment
The recurring crashes point to buggy firmware in the aaews service:

NULL Pointer Dereference: Attempt to read memory at 0x0 (common programming error).

Memory Corruption: Invalid addresses like 0xb65db170 suggest heap/stack corruption.

Possible Triggers:

Firmware bug in the AAE service (common in older ASUS firmware).

Conflict with a specific feature (e.g., AiCloud, remote access).

Memory leaks causing resource exhaustion.

gnuton, 2025-06-07:
Hi,
I checked the prebuild and they look fine, would you mind paste the logs? Chat GPT unfortunately does not help

hammerjoe, 2025-06-07:
Jun  7 19:54:33 kernel: potentially unexpected fatal signal 11.
Jun  7 19:54:33 kernel: CPU: 0 PID: 3754 Comm: aaews Tainted: P           O    4.1.52 #1
Jun  7 19:54:33 kernel: potentially unexpected fatal signal 11.
Jun  7 19:54:33 kernel: CPU: 1 PID: 3749 Comm: aaews Tainted: P           O    4.1.52 #1
Jun  7 19:54:33 kernel: Hardware name: Generic DT based system
Jun  7 19:54:33 kernel: task: d4c91c00 ti: ce9b8000 task.ti: ce9b8000
Jun  7 19:54:33 kernel: PC is at 0x0
Jun  7 19:54:33 kernel: LR is at 0x1acd8
Jun  7 19:54:33 kernel: pc : [<00000000>]    lr : [<0001acd8>]    psr: 60080010
Jun  7 19:54:33 kernel: sp : bebdf4f8  ip : b65e5b3c  fp : bebdf5d8
Jun  7 19:54:33 kernel: r10: 0001e03f  r9 : 6844c329  r8 : 0001e053
Jun  7 19:54:33 kernel: r7 : 00000041  r6 : 00031c66  r5 : 00031a08  r4 : 00031000
Jun  7 19:54:33 kernel: r3 : 00000000  r2 : 00000001  r1 : bebdf468  r0 : 00000000
Jun  7 19:54:33 kernel: Flags: nZCv  IRQs on  FIQs on  Mode USER_32  ISA ARM  Segment user
Jun  7 19:54:33 kernel: Control: 10c5387d  Table: 0aa7c04a  DAC: 00000015
Jun  7 19:54:33 kernel: CPU: 1 PID: 3749 Comm: aaews Tainted: P           O    4.1.52 #1
Jun  7 19:54:33 kernel: Hardware name: Generic DT based system
Jun  7 19:54:33 kernel: [<c0026f20>] (unwind_backtrace) from [<c0022cf8>] (show_stack+0x10/0x14)
Jun  7 19:54:33 kernel: [<c0022cf8>] (show_stack) from [<c045ba6c>] (dump_stack+0x8c/0xa0)
Jun  7 19:54:33 kernel: [<c045ba6c>] (dump_stack) from [<c003ab8c>] (get_signal+0x490/0x558)
Jun  7 19:54:33 kernel: [<c003ab8c>] (get_signal) from [<c0022290>] (do_signal+0xc8/0x3ac)
Jun  7 19:54:33 kernel: [<c0022290>] (do_signal) from [<c0022718>] (do_work_pending+0x94/0xa4)
Jun  7 19:54:33 kernel: [<c0022718>] (do_work_pending) from [<c001f58c>] (work_pending+0xc/0x20)
Jun  7 19:54:33 kernel: Hardware name: Generic DT based system
Jun  7 19:54:33 kernel: task: d6360c00 ti: ce9f4000 task.ti: ce9f4000
Jun  7 19:54:33 kernel: PC is at 0xb662a170
Jun  7 19:54:33 kernel: LR is at 0xb654db50
Jun  7 19:54:33 kernel: pc : [<b662a170>]    lr : [<b654db50>]    psr: 80080010
Jun  7 19:54:33 kernel: sp : b654cf28  ip : b654cf14  fp : b654d1cc
Jun  7 19:54:33 kernel: r10: 0002056d  r9 : 00020346  r8 : 00000000
Jun  7 19:54:33 kernel: r7 : 000000a2  r6 : 7fffffff  r5 : b654cf38  r4 : b654cf38
Jun  7 19:54:33 kernel: r3 : 00000000  r2 : 00000002  r1 : b654cf38  r0 : b654cf38
Jun  7 19:54:33 kernel: Flags: Nzcv  IRQs on  FIQs on  Mode USER_32  ISA ARM  Segment user
Jun  7 19:54:33 kernel: Control: 10c5387d  Table: 0aa7c04a  DAC: 00000015
Jun  7 19:54:33 kernel: CPU: 0 PID: 3754 Comm: aaews Tainted: P           O    4.1.52 #1
Jun  7 19:54:33 kernel: Hardware name: Generic DT based system
Jun  7 19:54:33 kernel: [<c0026f20>] (unwind_backtrace) from [<c0022cf8>] (show_stack+0x10/0x14)
Jun  7 19:54:33 kernel: [<c0022cf8>] (show_stack) from [<c045ba6c>] (dump_stack+0x8c/0xa0)
Jun  7 19:54:33 kernel: [<c045ba6c>] (dump_stack) from [<c003ab8c>] (get_signal+0x490/0x558)
Jun  7 19:54:33 kernel: [<c003ab8c>] (get_signal) from [<c00224b8>] (do_signal+0x2f0/0x3ac)
Jun  7 19:54:33 kernel: [<c00224b8>] (do_signal) from [<c0022718>] (do_work_pending+0x94/0xa4)
Jun  7 19:54:33 kernel: [<c0022718>] (do_work_pending) from [<c001f58c>] (work_pending+0xc/0x20)
Jun  7 19:54:45 kernel: potentially unexpected fatal signal 11.
Jun  7 19:54:45 kernel: CPU: 0 PID: 3777 Comm: aaews Tainted: P           O    4.1.52 #1
Jun  7 19:54:45 kernel: potentially unexpected fatal signal 11.
Jun  7 19:54:45 kernel: CPU: 1 PID: 3772 Comm: aaews Tainted: P           O    4.1.52 #1
Jun  7 19:54:45 kernel: Hardware name: Generic DT based system
Jun  7 19:54:45 kernel: task: ce946800 ti: ce9f4000 task.ti: ce9f4000
Jun  7 19:54:45 kernel: PC is at 0x0
Jun  7 19:54:45 kernel: LR is at 0x1acd8
Jun  7 19:54:45 kernel: pc : [<00000000>]    lr : [<0001acd8>]    psr: 60080010
Jun  7 19:54:45 kernel: sp : befcc4f8  ip : b65c1b3c  fp : befcc5d8
Jun  7 19:54:45 kernel: r10: 0001e03f  r9 : 6844c335  r8 : 0001e053
Jun  7 19:54:45 kernel: r7 : 00000041  r6 : 00031c66  r5 : 00031a08  r4 : 00031000
Jun  7 19:54:45 kernel: r3 : 00000000  r2 : 00000001  r1 : befcc468  r0 : 00000000
Jun  7 19:54:45 kernel: Flags: nZCv  IRQs on  FIQs on  Mode USER_32  ISA ARM  Segment user
Jun  7 19:54:45 kernel: Control: 10c5387d  Table: 0d58404a  DAC: 00000015
Jun  7 19:54:45 kernel: CPU: 1 PID: 3772 Comm: aaews Tainted: P           O    4.1.52 #1
Jun  7 19:54:45 kernel: Hardware name: Generic DT based system
Jun  7 19:54:45 kernel: [<c0026f20>] (unwind_backtrace) from [<c0022cf8>] (show_stack+0x10/0x14)
Jun  7 19:54:45 kernel: [<c0022cf8>] (show_stack) from [<c045ba6c>] (dump_stack+0x8c/0xa0)
Jun  7 19:54:45 kernel: [<c045ba6c>] (dump_stack) from [<c003ab8c>] (get_signal+0x490/0x558)
Jun  7 19:54:45 kernel: [<c003ab8c>] (get_signal) from [<c0022290>] (do_signal+0xc8/0x3ac)
Jun  7 19:54:45 kernel: [<c0022290>] (do_signal) from [<c0022718>] (do_work_pending+0x94/0xa4)
Jun  7 19:54:45 kernel: [<c0022718>] (do_work_pending) from [<c001f58c>] (work_pending+0xc/0x20)
Jun  7 19:54:45 kernel: Hardware name: Generic DT based system
Jun  7 19:54:45 kernel: task: cf059000 ti: d4cb6000 task.ti: d4cb6000
Jun  7 19:54:45 kernel: PC is at 0xb6606170
Jun  7 19:54:45 kernel: LR is at 0xb6529b50
Jun  7 19:54:45 kernel: pc : [<b6606170>]    lr : [<b6529b50>]    psr: 80080010
[... 437 lines cut ...]
Jun  7 19:56:57 kernel: [<c0026f20>] (unwind_backtrace) from [<c0022cf8>] (show_stack+0x10/0x14)
[... cut ...]
Jun  7 19:56:57 kernel: [<c0026f20>] (unwind_backtrace) from [<c0022cf8>] (show_stack+0x10/0x14)
Jun  7 19:56:57 kernel: [<c0022cf8>] (show_stack) from [<c045ba6c>] (dump_stack+0x8c/0xa0)
Jun  7 19:56:57 kernel: [<c045ba6c>] (dump_stack) from [<c003ab8c>] (get_signal+0x490/0x558)
Jun  7 19:56:57 kernel: [<c003ab8c>] (get_signal) from [<c00224b8>] (do_signal+0x2f0/0x3ac)
Jun  7 19:56:57 kernel: [<c00224b8>] (do_signal) from [<c0022718>] (do_work_pending+0x94/0xa4)
Jun  7 19:56:57 kernel: [<c0022718>] (do_work_pending) from [<c001f58c>] (work_pending+0xc/0x20)

gnuton, 2025-06-11:
I need to understand how you manage to trigger this issue, because I cannot reproduce it.
What do after resetting to factory the device?

closed 2025-07-17 by gnuton

gnuton, 2025-07-18:
Mostly debugging messages. This is a known issue that's harmless. Disable trendmicro and you will get rid of it.