gnuton/asuswrt-merlin.ng #789: RT-AX58U_v2 3004.388.9_2-gnuton1 beta 1 & 2

Derrick953, 2025-06-07:
Asus Router RT-AX58U V2

I'm having an issue with both the new beta releases in regards to the Aimesh.

I have uploaded both versions and then had to restore back to last stable version  3004.388.8_4-gnuton1.

Also then apply settings backup to get system back to where it works and is stable.

I not that great with logs and posting about the technical issues of what is wrong.

gnuton, 2025-06-07:
What is the problem? can you please elaborate?

Derrick953, 2025-06-08:
> have you tried to reset to factorty after flashing?

No, I have not. Upgrades have always worked without issue.

Kief5555, 2025-06-16:
I'm having the same issue - when trying to update the router, it fails then goes back to the original firmware before flashing the new one. I can still SSH to the router, however, it does not show up in the AIMesh page. 

`ASUSWRT-Merlin RT-AX82U_V2 3004.388.8_4-gnuton1 Sat Jan 11 06:16:27 UTC 2025`

gnuton, 2025-06-16:
@Kief5555

gnuton, 2025-06-19:
@Kief5555 the logs in System status http://router.asus.com/Main_LogStatus_Content.asp