gnuton/asuswrt-merlin.ng #803: [ZenWiFi XT8 / RT-AX95Q v1 - AI mesh not finding node  - firmware -RT-AX95Q_3004_388.9_2-gnuton1_puresqubi

mohammadpatel, 2025-06-25:
**Router Model Affected**
[ZenWiFi XT8 / RT-AX95Q v1

**Firmware Version Affected**
RT-AX95Q_3004_388.9_2-gnuton1_puresqubi

**Is this bug present in upstream Merlin releases too?**
no

**Describe the bug**
i updated to RT-AX95Q_3004_388.9_2-gnuton1_puresqubi, lost the AI mess functionality, i updated both router to this version, i tried connecting through wiredbackhall, ai mesh doesn't detect the node during search, i tried on app, and on router page.  it does assign an ip to the second router, but second router behaves as repeater, but ai mess doesn't seems to work (neither through wifi nor through wired backhall).

I had [3004.388.8_2-gnuton1](https://github.com/gnuton/asuswrt-merlin.ng/releases/tag/3004.388.8_2-gnuton1) before that was working fine, no issue

gnuton, 2025-06-25:
Please reset to factory

andrewk123, 2025-06-30:
I tried this today using a pair of XT8s running 3004_388.9_2-gnuton1 and was able to add Satellite nodes without any problem.  My main node was in "Router Mode" and I had "Ethernet Backhaul Mode" set to "On".

delaveganz, 2025-07-01:
Worked fine on my XT8 AI-mesh router with 3 XT8 nodes.