gnuton/asuswrt-merlin.ng #804: [RT-AX82U] Proceeds to upload, but does not change firmware

jhynzar, 2025-06-25:
**Router Model Affected**
Model: RT-AX82U v2

**Firmware Version Affected**
3004.388.9_2-gnuton1

**Is this bug present in upstream Merlin releases too?**
Yes/No

**Describe the bug**
Can't update the firmware from  stock 388_25035 and merlin 388.8_4

Coming from 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Administration' > 'Firmware Upgrade"
2. On 'Manual Firmware Update' Click on 'Upload'
3. Wait for the upload the finish
4. on reboot, Log in on the router, it displays the same previous firmware

**Expected behavior**
New firmware should be installed

Pheelix, 2025-06-25:
Same, I have a V2 and tried Beta 2 & 3 and it doesn't update.  The latest version I was able to update to was 3004.388.8_4-gnuton1.

witnwndr, 2025-06-26:
Same issues with V2

gnuton, 2025-06-26:
Can some of you paste the logs you see after upgrading the firmware? do you see kernel crashes?

gnuton, 2025-06-26:
@Fire2fly please file that in a new ticket since it's a different model and the problem is most probably gonna be different.

gnuton, 2025-07-08:
Hey there,
can you please test this firmware https://github.com/gnuton/asuswrt-merlin.ng/actions/runs/16140629361

presterjohn13, 2025-07-09:
> Hey there, can you please test this firmware https://github.com/gnuton/asuswrt-merlin.ng/actions/runs/16140629361

This one has still has the same issue for me (82U V2)

Sorry gnuton

I appreciate all your work on this

gnuton, 2025-07-09:
Thanks a lot @Mive82 so the fix worked out. this is a new issue -_-. Step2

gnuton, 2025-07-17:
Fix here https://github.com/gnuton/asuswrt-merlin.ng/releases/tag/3004.388.9_2-gnuton2_beta1

closed 2025-07-17 by gnuton