gnuton/asuswrt-merlin.ng #815: TUF-AX3000 V2 as ai mesh node no wifi can connect

kanicrab, 2025-07-03:
**Router Model Affected**
Models: (eg: DSL-AX82U/RT-AX82U/...)
TUF-AX3000 V2

**Firmware Version Affected**
3004.388.9_2-gnuton1

**Is this bug present in upstream Merlin releases too?**
No

**Describe the bug**
A clear and concise description of what the bug is.
No wireless client can connect to the node, even bind the client to the node.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem or remove this section

yo3gts, 2025-07-03:
Same thing here at RT-AX58U, before with the old firmware RT-AX58U_V2_3004_388.8_4-gnuton1 it worked.
Now no client can connect to AiMesh Node, only via cable

yo3gts, 2025-07-09:
same thing here...after revert old firmware, works...

gnuton, 2025-07-17:
@PAGuyWithABlueSky it would be great to get the spanning tree issue better described in a github issue

Harvester57, 2025-09-21:
As a rule of thumb, nodes should not use Merlin or Gnuton, and stay on the official firmware release. There is no added value in using Gnuton/Merlin on nodes anyway.

uscmbrixton, 2025-11-21:
I have not tested again but I believe this is the same issue as #891 and the issue is most likely still there.