gnuton/asuswrt-merlin.ng #825: 3004.388.9_2-gnuton1 nodes freeze every 1187 seconds

troymuller, 2025-07-13:
**Router Model Affected**
XT8

**Firmware Version Affected**
3004.388.9_2-gnuton1

**Is this bug present in upstream Merlin releases too?**
Unknown

**Describe the bug**
Approximately every 1187 seconds, my node in my AI mesh (wireless) will simply drop traffic for approximately 10-15 seconds.

**To Reproduce**
Unsure.

**Expected behavior**
Traffic not dropping

I applied the 3004.388.9_2-gnuton1 firmware, did a factory reset and re-setup my XT8 devices with the app.  Switched from router to AP mode and also turned off smart connect.  

Clients seem to maintain access to node device but node device simply starts dropping traffic.  There isn't anything obvious in the logs when this occurs but I am more than willing to provide both those and ping traces to show the behavior (I don't have them right this second).

I have looked over every config item to see if there was anything that would bring about this, whether a re-key was going on at the time, etc to no avail.

Any thoughts as to what could be causing this?  

Setup is main node is connected via 2.5Gbps port to my network and wirelessly connected to the nodes in AP mode.

Thank you in advance.

troymuller, 2025-07-13:
Here's a snippet of the issue in action pinging from a host attached to the node.

64 bytes from 192.168.1.7: icmp_seq=65 ttl=64 time=2.328 ms
Request timeout for icmp_seq 66
64 bytes from 192.168.1.7: icmp_seq=66 ttl=64 time=1097.997 ms
Request timeout for icmp_seq 68
64 bytes from 192.168.1.7: icmp_seq=67 ttl=64 time=2187.770 ms
Request timeout for icmp_seq 70
Request timeout for icmp_seq 71
Request timeout for icmp_seq 72
Request timeout for icmp_seq 73
Request timeout for icmp_seq 74
Request timeout for icmp_seq 75
Request timeout for icmp_seq 76
64 bytes from 192.168.1.7: icmp_seq=76 ttl=64 time=1672.220 ms
64 bytes from 192.168.1.7: icmp_seq=77 ttl=64 time=670.351 ms

Host attached to the main node never lost ping to that main node, only a host on a satellite seems to see this.

gnuton, 2025-07-14:
Hi @troymuller 

The system logs you attached are particularly interesting. The lines Jul 13 19:35:33 watchdog: detect wl reinit count 18 and Jul 13 19:35:33 watchdog: reinit of unit2:18 indicate that the wireless interface (wl) on that satellite node is restarting. This is exactly the kind of behavior that could cause the drops and timeouts you're seeing in your pings. The "watchdog" is a mechanism that monitors the system, and when it detects a persistent problem or a crash, it attempts to restart a component to restore functionality.

**Possible Causes for Wireless Interface Reinitialization**
There could be several reasons for this wireless interface reinitialization, including:

- Wireless Driver Issues: There might be a bug in the wireless driver for that specific hardware, leading to crashes or instability under certain conditions.

- Hardware Problems: While less common, a hardware defect on the satellite's wireless module could cause instability.

- Environmental Interference: Even though you mentioned a good signal, strong interference from other wireless networks or devices in the same area could destabilize the connection.

- Node Overload: If the satellite node is under unusually heavy load (too many clients, high traffic), it might struggle to maintain stability.

**Information Needed for Further Diagnosis**
To help us diagnose this further, could you please provide the following information?

- Exact Asuswrt-Merlin firmware version you're running on all nodes (main router and XT8 satellites).
- 
- How the problematic satellite is connected: Is it wirelessly backhauled (as it seems to indicate) or is there a wired backhaul we're unaware of?
- 
- Any custom configurations you might have applied to the affected satellite node (e.g., specific wireless channel settings, bandwidth, etc.).
- 
- If possible, try relocating the problematic satellite node to a different position (perhaps temporarily closer to the main node) to see if the issue persists. This could help us rule out environmental or interference-related problems.
- 
- Check the AiMesh backhaul status for that satellite node through the main router's web interface. See if there are any indications of issues or changes in backhaul signal quality.

In the meantime, you might try a complete power cycle (turning off and on again) of both the problematic satellite node and the main node. Sometimes, a simple reboot can resolve temporary glitches.

troymuller, 2025-07-15:
Information Needed for Further Diagnosis
To help us diagnose this further, could you please provide the following information?

Exact Asuswrt-Merlin firmware version you're running on all nodes (main router and XT8 satellites).

I have seen this behavior exhibited on 3004.388.9_2-gnuton1 and Stock firmware 3.0.0.4.388_24709

How the problematic satellite is connected: Is it wirelessly backhauled (as it seems to indicate) or is there a wired backhaul we're unaware of?

Wireless backhaul only.   The main node is the only one with a cable attached to it.

Any custom configurations you might have applied to the affected satellite node (e.g., specific wireless channel settings, bandwidth, etc.).

5Ghz-2:

20/40/80/160 Mhz, Channel 165, group rotation 14400, smart connect disabled, AX only wireless mode.   The rest are mainly default with the exception of disabling universal beam forming.

If possible, try relocating the problematic satellite node to a different position (perhaps temporarily closer to the main node) to see if the issue persists. This could help us rule out environmental or interference-related problems.

Relocated the satellite next to the main node and the problem persists.

Check the AiMesh backhaul status for that satellite node through the main router's web interface. See if there are any indications of issues or changes in backhaul signal quality.

During an event, no indication of signal quality changing.

troymuller, 2025-07-18:
Surprisingly, a similar thing happens on a wired backhaul connection, I am not sure why, this seems strange to me, why this would affect both wireless and wired connections.  Something must be monitoring network connections and deciding to reset them for some reason or another, that is what I can't seem to find anywhere in logs.

gnuton, 2025-07-19:
please check if you have this bug with stock firmware.
it would also be interesting to see logs and you may also check resource (cpu usage with top)

troymuller, 2025-07-19:
for the wireless portion, I found this as previously denoted in a prior post.

I have seen this behavior exhibited on 3004.388.9_2-gnuton1 and Stock firmware 3.0.0.4.388_24709 on wireless backhaul.

veepee78, 2025-07-19:
I'll add useless comment, I believe I have same issue. Using 2 x xt8 with 9_2 gnuton1. Did not investigate this deeply, just noticed that video playback was hanging quite constant intervals. Wasn't sure if it was wifi or 5g, so I tested another wlan router as ap. Problem seemed to go away.

I'm getting feeling that these xt8's are nothing but trouble. I believe I have to try some other hardware.

Thank you gnuton for at least trying to fix these things.

czyz, 2026-01-11:
I'm seeing similar behavior in both 3004.388.9_2-gnuton2 and 3004_388.10_2-gnuton1, _not only when connected to nodes (I have 2 connected via wired backhaul) via wifi_, but even when just using wired ethernet to my main XT8 router. The connection just disappears on what seems like a fairly regular schedule (mostly just under 19 minutes of connectivity before just under a minute of the router being non-communicative before it returns to normal service). 

I can't do much diagnostic checking during the outages because my ssh connection to the router becomes non-responsive, both via the web UI and ssh. If I leave it the router pinging google, I can see a series of pings returned at normal speed for those 19 minutes and then it gets longer durations for a while before returning back to normal timing. 

64 bytes from 142.251.142.238: seq=1059 ttl=105 time=161.493 ms
64 bytes from 142.251.142.238: seq=1060 ttl=105 time=161.431 ms
64 bytes from 142.251.142.238: seq=1061 ttl=105 time=161.959 ms
64 bytes from 142.251.142.238: seq=1062 ttl=105 time=161.966 ms
64 bytes from 142.251.142.238: seq=1063 ttl=105 time=161.503 ms
64 bytes from 142.251.142.238: seq=1064 ttl=105 time=161.573 ms
64 bytes from 142.251.142.238: seq=1065 ttl=105 time=761.272 ms
64 bytes from 142.251.142.238: seq=1066 ttl=105 time=609.998 ms
64 bytes from 142.251.142.238: seq=1067 ttl=105 time=7132.363 ms
64 bytes from 142.251.142.238: seq=1068 ttl=105 time=15730.868 ms
64 bytes from 142.251.142.238: seq=1069 ttl=105 time=14741.492 ms
64 bytes from 142.251.142.238: seq=1070 ttl=105 time=13755.233 ms
64 bytes from 142.251.142.238: seq=1071 ttl=105 time=12736.905 ms
64 bytes from 142.251.142.238: seq=1072 ttl=105 time=11742.707 ms
64 bytes from 142.251.142.238: seq=1073 ttl=105 time=10750.464 ms
64 bytes from 142.251.142.238: seq=1074 ttl=105 time=9765.384 ms
64 bytes from 142.251.142.238: seq=1075 ttl=105 time=8791.076 ms
64 bytes from 142.251.142.238: seq=1076 ttl=105 time=7801.445 ms
64 bytes from 142.251.142.238: seq=1077 ttl=105 time=6678.103 ms
64 bytes from 142.251.142.238: seq=1078 ttl=105 time=9386.141 ms
64 bytes from 142.251.142.238: seq=1079 ttl=105 time=8399.112 ms
64 bytes from 142.251.142.238: seq=1084 ttl=105 time=3451.757 ms
64 bytes from 142.251.142.238: seq=1085 ttl=105 time=2502.615 ms
64 bytes from 142.251.142.238: seq=1086 ttl=105 time=1768.727 ms
64 bytes from 142.251.142.238: seq=1087 ttl=105 time=803.460 ms
64 bytes from 142.251.142.238: seq=1088 ttl=105 time=163.753 ms
64 bytes from 142.251.142.238: seq=1089 ttl=105 time=164.982 ms
64 bytes from 142.251.142.238: seq=1090 ttl=105 time=161.947 ms
64 bytes from 142.251.142.238: seq=1091 ttl=105 time=162.125 ms
64 bytes from 142.251.142.238: seq=1092 ttl=105 time=167.052 ms
64 bytes from 142.251.142.238: seq=1093 ttl=105 time=364.789 ms
64 bytes from 142.251.142.238: seq=1094 ttl=105 time=229.794 ms
64 bytes from 142.251.142.238: seq=1095 ttl=105 time=367.425 ms
64 bytes from 142.251.142.238: seq=1096 ttl=105 time=162.841 ms
64 bytes from 142.251.142.238: seq=1097 ttl=105 time=161.484 ms
64 bytes from 142.251.142.238: seq=1098 ttl=105 time=172.616 ms
64 bytes from 142.251.142.238: seq=1099 ttl=105 time=161.572 ms
64 bytes from 142.251.142.238: seq=1100 ttl=105 time=161.497 ms
64 bytes from 142.251.142.238: seq=1101 ttl=105 time=161.498 ms
64 bytes from 142.251.142.238: seq=1102 ttl=105 time=161.493 ms

On my computer connected to the router (via ethernet), when the router goes into this weird state the result of a ping test looks like: 

12:22:44.622142 64 bytes from 142.251.142.238: icmp_seq=3913 ttl=104 time=160.359 ms
12:22:45.623180 64 bytes from 142.251.142.238: icmp_seq=3914 ttl=104 time=160.776 ms
12:22:46.627843 64 bytes from 142.251.142.238: icmp_seq=3915 ttl=104 time=160.602 ms
12:22:54.855421 64 bytes from 142.251.142.238: icmp_seq=3916 ttl=104 time=7384.124 ms
12:22:56.038800 64 bytes from 142.251.142.238: icmp_seq=3917 ttl=104 time=7562.483 ms
12:23:14.562539 Request timeout for icmp_seq 3942
12:23:15.563711 Request timeout for icmp_seq 3943
12:23:16.569260 Request timeout for icmp_seq 3944
12:23:16.871326 64 bytes from 142.251.142.238: icmp_seq=3936 ttl=104 time=9328.082 ms
12:23:18.577547 Request timeout for icmp_seq 3946
12:23:18.839113 64 bytes from 142.251.142.238: icmp_seq=3937 ttl=104 time=10293.157 ms
12:23:20.545823 64 bytes from 142.251.142.238: icmp_seq=3938 ttl=104 time=10994.819 ms
12:23:21.587945 Request timeout for icmp_seq 3949
12:23:22.588550 Request timeout for icmp_seq 3950
12:23:23.591542 Request timeout for icmp_seq 3951
12:23:24.258691 64 bytes from 142.251.142.238: icmp_seq=3939 ttl=104 time=13706.864 ms
12:23:25.595927 Request timeout for icmp_seq 3953
12:23:26.601122 Request timeout for icmp_seq 3954
12:23:27.486625 64 bytes from 142.251.142.238: icmp_seq=3940 ttl=104 time=15933.161 ms
12:23:28.607630 Request timeout for icmp_seq 3956
12:23:29.612681 Request timeout for icmp_seq 3957
12:23:30.469409 64 bytes from 142.251.142.238: icmp_seq=3941 ttl=104 time=17914.374 ms
12:23:31.622191 Request timeout for icmp_seq 3959
[... 89 lines cut ...]
18:52:34  00:18:51  ✓ Good connectivity (ping < 900ms)
19:11:44  00:00:36  ✗ Poor connectivity (timeout or ping >= 900ms)
19:12:21  00:18:53  ✓ Good connectivity (ping < 900ms)
19:31:32  00:00:35  ✗ Poor connectivity (timeout or ping >= 900ms)
19:32:08  00:18:57  ✓ Good connectivity (ping < 900ms)
19:51:15  00:00:45  ✗ Poor connectivity (timeout or ping >= 900ms)
[... cut ...]
21:11:20  00:13:15  ✓ Good connectivity (ping < 900ms)

Note: 00:00:00 durations indicate single-ping spans (connectivity changed immediately)

Summary:
--------
  Total good connectivity: 3h 55m 5.99s (95.71%)
  Total poor connectivity: 10m 32.18s (4.28%)
  Total time analyzed: 4h 5m 38.16s

This may be totally unrelated to the original problem in this thread. But the pings looked familiar. I see no watchdog messages in the syslog or anything that seems likely to be related to the issue (as far as I can tell).