gnuton/asuswrt-merlin.ng #828: RT-AX58U V2 Icloud issue

Masterload1975, 2025-07-18:
Still issue since 3004_388.9_1

My SSD is empty and if you click on it:

Server error! Please refresh page.

(No issue in 3004_388.8_4)

[image]

gnuton, 2025-07-19:
Do you have any log to share?

gnuton, 2025-07-20:
icloud is a service that may be deactivated in the future, given some security issues it may have.
This said, if you do nto see anyhing in the log you may try to ssh ,run a ps , find out the process runnig, then you can stop and restarting it. You will get logs just right in the terminal.. 
I never tried this, but with some deaemons this works