gnuton/asuswrt-merlin.ng #829: skynet doesn't work to 3004.388.9_2-gnuton2_beta1

antoniosgkogk-ux, 2025-07-20:
**Router Model Affected**
Models: (TUF-AX3000 V2)

**Firmware Version Affected**
3004.388.9_2-gnuton2_beta1

**Is this bug present in upstream Merlin releases too?**
Yes/No

**Describe the bug**
IPSets and IPTables Rules failing to load within latest version but in [3004.388.8_4-gnuton1] works just fine
**To Reproduce**
Steps to reproduce the behavior:
Simple install Skynet and you will see it 

[image]

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem or remove this section

YakudzaKY, 2025-07-29:
+1 on RT-AX58U V2
before beta xt_proxy was working, now both broken

Pochatok@RT-AX58U-5E40:/tmp/home/root# modprobe xt_set
modprobe: FATAL: Module xt_set not found in directory /lib/modules/4.19.183
Pochatok@RT-AX58U-5E40:/tmp/home/root# modprobe xt_tproxy
modprobe: FATAL: Module xt_tproxy not found in directory /lib/modules/4.19.183

zebragr, 2025-09-04:
Resolved for RT-AX58U V2 with latest release:  3004.388.9_2-gnuton2