gnuton/asuswrt-merlin.ng #832: Skynet Firmware Bug?

ScruffySpaceCowboy, 2025-07-23:
Same issue as https://github.com/gnuton/asuswrt-merlin.ng/issues/559.

Clean install with latest firmware on TUF-AX3000v2, latest entware dir, latest amtm.

Skynet will install but will fail IPSets and IPTables Rules with Rule Integrity Violations. (see attached)

[image]
[image]

gnuton, 2025-07-24:
Same as #787 even if a different model

gnuton, 2025-07-24:
Test images here https://github.com/gnuton/asuswrt-merlin.ng/actions/runs/16487789919
fix: https://github.com/gnuton/asuswrt-merlin.ng/commit/01c0423dd97bfd520083e867bcc9af9ea76e9faa

ScruffySpaceCowboy, 2025-07-24:
> Test images here https://github.com/gnuton/asuswrt-merlin.ng/actions/runs/16487789919 fix: [01c0423](https://github.com/gnuton/asuswrt-merlin.ng/commit/01c0423dd97bfd520083e867bcc9af9ea76e9faa)

These all have the same EACESS permission denied artifact.
Does that impinge on the validity of the builds?

gnuton, 2025-07-25:
@ScruffySpaceCowboy have you had time to test the images?

closed 2025-07-25 by ScruffySpaceCowboy