gnuton/asuswrt-merlin.ng #833: Unable to revert to original firmware - AX-82U

mitheshge, 2025-07-27:
**Router Model Affected**
RT-AX82U

**Firmware Version Affected**
Current Version : 3004.388.9_2-gnuton1

**Is this bug present in upstream Merlin releases too?**
Yes/No

**Describe the bug**
Unable to restore/revert to the original firmware. It says firmware upgrade unsuccessful. (Screesnshots attached)

[image]
[image]

HarryDotMYx, 2025-07-29:
Hi there, please read #757

HarryDotMYx, 2025-07-29:
Hi, thanks for the details. According to ASUS’s official guide, here’s what to do:

1. Set your PC’s IPv4 manually to 192.168.1.10, subnet 255.255.255.0 — do not use DHCP while in rescue mode.
2. Power off your router, then press and hold the Reset button, plug in the power and keep holding until the power LED blinks slowly — that indicates rescue mode is active.
3. Launch the ASUS Firmware Restoration tool, select your firmware file, and click Upload once the LED is blinking slowly. Wait until the LED turns solid before closing the utility.

If the tool still says “not in rescue mode,” check that you’re using the exact static IP 192.168.1.10 and not relying on Device Discovery IPs. Also, disable firewall/virtual-net adapters, connect directly via LAN port 1, and try another PC or cable if possible.

Let me know how it goes.

mitheshge, 2025-07-29:
Got it to work. 

The above IP as mentioned didn't work as my router comes with the private IP 192.168.50.1. So I just used 50.2 as my IPV4 for my laptop. 
Also, All the network adaptors should be disabled except the ethernet Adaptor. Only then the tool works. 

Firmware restored to stock(latest).

Thanks for the help.

HarryDotMYx, 2025-07-30:
Great to hear it worked!

Thanks for confirming the fix — this is really helpful:

Setting your laptop’s IP to 192.168.50.2 to match the router’s 192.168.50.1 was the right move.

Disabling all other network adapters except the Ethernet adapter is also an important step — especially for the ASUS Restoration Tool to detect the router properly.

Glad you got the firmware restored to stock successfully!

Appreciate you sharing the solution — this will definitely help others in the same situation.

And please don't forget to close this issues!

agibson-fl, 2026-09-16:
> If the tool still says “not in rescue mode,” check that you’re using the exact static IP 192.168.1.10 and not relying on Device Discovery IPs. Also, disable firewall/virtual-net adapters, connect directly via LAN port 1, and try another PC or cable if possible.
> 
> Let me know how it goes.

I just spend hours with this, until I saw your comment/suggestion. I could not disable the Hyper-V adapter (disable was greyed out). then I got a (i think) brilliant idea!  I turned off the checkboxes for ipv4 and ipv6 inside the Hyper-V adapter, and boom, the restore app started uploading immediately! Thanks. Hope this helps someone in the future.

Also, for me, the router was using 192.168.50.1 after I factory reset it, but restore mode still required me to use 192.168.1.10 as the ip address of the computer.