gnuton/asuswrt-merlin.ng #842: TUF-AX5400 V1: 'TUF-AX5400_3004_388.9_2-gnuton2_puresqubi.w' cause pages broken or missing tab.

mjamil85, 2025-08-09:
**Router Model Affected**
Models: TUF-AX5400 V1

**Firmware Version Affected**
3004.388.9_2-gnuton2

**Is this bug present in upstream Merlin releases too?**
Yes

**Describe the bug**
The bugs still the same after perform hard factory reset & manually configure setup (without upload backup setting). 

**Process**
 First try: Same Issue
- Update new firmware (TUF-AX5400_3004_388.9_2-gnuton2_puresqubi.w) > Perform Factory Reset > Create Manual Setup (without upload backup setting).

 Second try: Same Issue
- Update 2nd time new firmware (TUF-AX5400_3004_388.9_2-gnuton2_puresqubi.w) > Perform Hard Factory Reset (WPS + Power Button) > Create Manual Setup (without upload backup setting).

**Expected behavior**
Previous version **(TUF-AX5400_3004_388.8_4-gnuton1_tuf_puresqubi.w)** don't have this kind of issue but with new firmware (without TUF theme), cause this pages broken.

**Screenshots**

**Game Page**

[image]

**Wireless Time Schedule Page**

[image]

[image]

**Network Tools Page**

[image]

bobyang3, 2025-08-21:
confirmed. I have the same issue too

mrkaktuz, 2025-08-26:
Same problem tuf-ax3000 v2

Sp0Xik, 2025-09-02:
Same problem tuf-ax5400 V1

gnuton, 2025-11-23:
@mjamil85 thanks for confirming this works. I will close this issue.
The speedtest is a closed source binary from ASUS. The different results from the speed tests depends on the servers used for the tests. If you use the servers inside your ISP network of course those will be faster.
The results are in any case correct, but you have to know how to interpret the numbers.