gnuton/asuswrt-merlin.ng #844: ##  GNUton's Asus Merlin changelog  ##

jaca2300, 2025-08-13:
##  GNUton's Asus Merlin changelog  ##

Date: June 11, 2025
Release: 3006.102.4_1-gnuton1

  - ADEED: Support for GT-BE98

<hr /><em>This discussion was created from the release <a href='https://github.com/gnuton/asuswrt-merlin.ng/releases/tag/3006.102.4_1-gnuton1'>3006.102.4_1-gnuton1</a>.</em>

_Originally posted by @gnuton in https://github.com/gnuton/asuswrt-merlin.ng/discussions/794_

jaca2300, 2025-08-13:
Witch firmware should I use on my Rog GT-BE98: GT-BE98_3006_102.4_1-gnuton1_rog_nand_squashfs.pkgtb or
GT-BE98_3006_102.4_1-gnuton1_nand_squashfs.pkgtb
I don't think there is a non-rog GT-BE98

Neoony, 2025-08-13:
> Witch firmware should I use on my Rog GT-BE98: GT-BE98_3006_102.4_1-gnuton1_rog_nand_squashfs.pkgtb or
> GT-BE98_3006_102.4_1-gnuton1_nand_squashfs.pkgtb
> I don't think there is a non-rog GT-BE98 

well atleast for me, I used the rog one and it works fine
I assume its mostly just the theme of the UI

closed 2025-11-08 by gnuton