gnuton/asuswrt-merlin.ng #846: RT-AX58U V2 - Missing Advanced Menu/Switch button

r00tmebaby, 2025-08-15:
**Router Model Affected**
Models:  RT-AX58U V2

**Firmware Version Affected**
384.xx

**Is this bug present in upstream Merlin releases too?**
Yes

**Describe the bug**
Do not see menu tab on the left side. Tried to zoom in-out to see if advanced mode button appears. Did not fix it. 

I can still access and see the content if I hit the url http://192.168.50.1/Advanced_Wireless_Content.asp and http://192.168.50.1/Advanced_LAN_Content.asp but the menu on the left is still not present.

**To Reproduce**
Steps to reproduce the behavior:
Using default router mode
1. Just login and you land in simple UI, no option to switch.

**Expected behavior**
I expected to be able to use the firwamre, right now it provides less than the original :)

**Screenshots**

[image]

Facsimiler, 2025-08-25:
Same for me on ZenWifi AX (6600/XT8).

Facsimiler, 2025-08-25:
I did a factory reset of my router, and it came back with the main menu enabled, in case that helps.

gnuton, 2025-11-07:
Fixed in the latest release

closed 2025-11-07 by gnuton