gnuton/asuswrt-merlin.ng #854: Missing ASUS DDNS

micron888, 2025-09-05:
TUF 5400 v1
3004.388.9_2-gnuton2

ASUS dns is missing

[image]

alfky12, 2025-09-05:
Try this, while waiting for Gnuton's. It's based on Gnuton firmware with some fixes.
For TUF-AX5400 (v1) ONLY!
[Default UI](https://github.com/alfky12/asuswrt-merlin.ng/actions/runs/17263812726/artifacts/3862564236)
[TUF UI](https://github.com/alfky12/tuf-asuswrt-merlin.ng/actions/runs/17186963498/artifacts/3837429042)

micron888, 2025-09-08:
This build fix ddns and GUI issue report on #822

gnuton, 2025-10-18:
This feature seems to be disabled in some router models by ASUS.
fix here 252dfebdcad2ba6ba6edca0e9175a6e6ca1f2658

gnuton, 2025-11-08:
Fix part of the latest pre-release

closed 2025-11-08 by gnuton